m5stack/M5StickC #2: Deep Sleep and Display

bert2002, 2019-04-28:
Hi,
it seems that the deep sleep mode does not shut down the display. When using following code the display stays on:

```
esp_sleep_enable_timer_wakeup(60000000);
esp_deep_sleep_start();
```
Is there anything we can do about?

bert

bert2002, 2019-04-28:
I see that I can disable the display while booting, but even then the backlight gets on.

Reference: https://docs.m5stack.com/#/en/api/system_m5stickc

bert2002, 2019-04-28:
Mhmm seems that Wifi also does not get disabled, even if I use this:
```
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
```

EeeeBin, 2019-04-28:
Display Lcd is control by axp192, ldo2, Try call this fun https://github.com/m5stack/M5StickC/blob/master/src/AXP192.cpp#L372

bert2002, 2019-04-30:
How do I call this @EeeeBin ?

EeeeBin, 2019-04-30:
```M5.Axp.SetSleep()``` Called in front of deep sleep

bert2002, 2019-05-05:
Seems my library does not have the SetSleep() function, but I got it working with this:
```
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
```

closed 2019-05-05 by bert2002

anthonykeane, 2019-05-30:
> `M5.Axp.SetSleep()` Called in front of deep sleep

is it possible to wake from this with a button press or timer?

bert2002, 2019-06-01:
> > `M5.Axp.SetSleep()` Called in front of deep sleep
> 
> @bert2002 is it possible to wake from this with a button press or timer?

I didn't try it, but I saw an implementation in the M5Stack Firmware, which was not available in the M5StickC implementation. Then I stopped looking into it.

jumpjack, 2020-05-25:
M5.Lcd.writecommand(ST7735_DISPOFF); // Turn off screen, but not backlight
(https://community.m5stack.com/topic/94/turn-off-display/6 )

  M5.Axp.ScreenBreath(0); // Turn off backlight