m5stack/M5StickC #9: errors compiling HelloWorld

Bluebrain2000, 2019-05-24:
Windows, Android 1.8.9
compiling the HelloWorld Example gives me the following errors:

Arduino\libraries\M5StickC\src\RTC.cpp: In member function 'void RTC::Bcd2asc()':
Arduino\libraries\M5StickC\src\RTC.cpp:73:24: error: suggest parentheses around arithmetic in operand of '|' [-Werror=parentheses]

Arduino\libraries\M5StickC\src\M5Display.cpp: In member function 'bool M5Display::initHzk16(boolean, const char*, const char*)':
Arduino\libraries\M5StickC\src\M5Display.cpp:115:10: error: enumeration value 'ExternalHzk16' not handled in switch [-Werror=switch]

Arduino\libraries\M5StickC\src\M5Display.cpp: At global scope:
Arduino\libraries\M5StickC\src\M5Display.cpp:441:1: error: 'include' does not name a type
 include "utility/qrcode.h"

[compiler_output.txt](https://github.com/m5stack/M5StickC/files/3217612/compiler_output.txt)

zhouyangyale, 2019-05-31:
Is the problem with the library version, you update it.

Bluebrain2000, 2019-05-31:
It's already the latest version.

TonyVie, 2019-06-26:
Same problem here.
(latest lib, latest arduino IDE, OSX)

In M5Display::initHzk16, added:
default:break;
but missing case should be addressed.

In RTC::Bcd2asc added some parentesis:
asc[j++] =(trdata[i]&0x0f)|0x30;

Zontex, 2020-11-12:
Library updated, issue should be solved.

closed 2020-11-12 by Zontex