m5stack/M5StickC #18: Please add M5StickC display support to esphome

apolselli, 2019-06-22:
Hello. I am using a couple of M5StackC devices as esphome nodes, with sensors attached, deep_sleep, buttons, IR transmitter, etc.. Here’s one of my configs 
https://github.com/apolselli/home-assistant-config/blob/master/esphome/m5stickc.yaml

It’s a pity that I still can’t use the internal display to show some sensors’ or system info.

Please add support for that display in esphome (a wonderful lib to integrate esp32 devices in home assistant). It should be very essy since the arduino code is already there... And of course, other components are welcome too.

Thank You

apolselli, 2019-06-22:
TFT LCD
Driver IC：ST7735S
Resolution：80 * 160
https://github.com/m5stack/M5-Schematic/blob/master/Core/ST7735S_v1.1.pdf

apolselli, 2019-06-22:
Please consider that esphome can become a killer application for M5StickC, when most of the components will be integrated.
Everyone wants external display, buttons and sensors for hassio / home assistant. Home assistant has millions of users. Esphome can act as a glue between esp prototyping boards and home automation enthusiasts. With no soldering required.

And they lived happily ever after.

aenertia, 2019-06-23:
Have a look at my cobbled together hack at putting a display on the WLED project with the m5stickC: 

https://github.com/aenertia/m5stickCLED/tree/master

zhouyangyale, 2019-06-25:
Hello, we will join later

xppx99, 2019-09-19:
Hello! I also have a few m5sticksC, for the money they cost they are of great value! 
I'm looking forward to have it's screen suported in ESPHome

tlrobinson, 2019-11-25:
Seconding, M5StickC and ESPHome seem like a perfect combination.

davidtgbe, 2020-01-13:
+1. It Would be great!

giuseppegimo, 2020-01-26:
I too am really interested in that. I bought an m5 stick and I hope to use it with esphome.

fdev31, 2020-01-30:
I am very interested as well.
With some guidance and/or time I could eventually use my coding skills on this. It's not the kind of task I already did and I'm new to ESPHome (a couple of days). I never wrote code to interface with a screen, only very basic devices, else I'm using libraries: if some developers can provide advices and/or code I could adapt it would be appreciated :)

airy10, 2020-02-13:
You can check https://github.com/airy10/esphome-m5stickC (still a work in progress)

apolselli, 2020-02-16:
It works like a charm! Great job

[image]

giuseppegimo, 2020-02-16:
Hi @apolselli . Could you please share your m5stick_thermostat code? thanks!

davidtgbe, 2020-02-27:
Works perfect. I've had to copy arial.tft fonts to esphome folder. I've BLE module loaded and it is associated to a Xiaomi temp and humidity sensor (https://esphome.io/components/sensor/xiaomi_lywsdcgq.html). I'm trying to use it as an wireless thermostat to drive my boiler.
The only issue I'm experiencing is that mi5stickc seems to be restarting periodically. Display goes off for less than a second and then restarts.

abmantis, 2020-03-08:
I've noticed that the IR stays ON after transmitting. Any tips on that?

xppx99, 2020-03-14:
> I've noticed that the IR stays ON after transmitting. Any tips on that?

I also can reproduce this problem.

sasanasit, 2020-04-16:
> You can check https://github.com/airy10/esphome-m5stickC (still a work in progress)

is there a way to turn the display on and off?
thank you

cromulus, 2020-07-24:
@airy10 +1 to turning display off. I want to use deep sleep mode, but the display is killing my battery life.

It appears that there is a display off command, which is different than the backlight (brightness)

Thanks!

ejdan, 2020-10-05:
> I've noticed that the IR stays ON after transmitting. Any tips on that?

You need to invert the transmitter:

```
remote_transmitter:
  pin:
    number: GPIO9
    inverted: true
  carrier_duty_percent: 50%
```

Zontex, 2020-11-16:
Seems like ESPHome support added by the community member, closing this issue for now

closed 2020-11-16 by Zontex

DougManuel, 2024-01-14:
Suggestion for disabling backlight:

https://github.com/martydingo/esphome-axp192/pull/7