m5stack/M5StickC #20: turn off display

wanleg, 2019-07-02:
the function `M5.Lcd.setBrightness(0)` returns the following when compiling:
```
C:\Users\user\Documents\Arduino\M5stickC_rtc/M5stickC_rtc.ino:55: undefined reference to `M5Display::setBrightness(unsigned char)'

collect2.exe: error: ld returned 1 exit status
```
I'm using `M5.Axp.SetSleep()` as a workaround for now, but it seems inconsistent when the rest of the functions listed [here](https://github.com/m5stack/M5Stack/blob/master/src/M5Stack.h) work fine.

wanleg, 2019-07-02:
sorry - meant to say [here](https://github.com/m5stack/M5StickC/blob/master/src/M5Display.h)

zhouyangyale, 2019-07-03:
void  ScreenBreath(uint8_t brightness);   
You can use this function

wanleg, 2019-07-03:
Thanks! I found that later, too. Hard to track all the functions when everything's under active development!

closed 2019-07-03 by wanleg

barabanus, 2019-12-09:
It looks like there's still no function to turn off display.
- LCD::sleep() and LCD::setBrightness() are not defined
- LCD::ScreenBreath() does not exist

jhoughjr, 2019-12-16:
this should maybe be reopened

sergeymaysak, 2020-01-04:
@barabanus 
M5.Axp.ScreenBreath(0); works for me well.

barabanus, 2020-01-05:
@sergeymaysak 
The display still takes a lot of power in this mode.

c0qu1, 2020-04-02:
Please fix the M5.Lcd.setBrightness(brightness) function. This needs to be reopened.

victordiaz, 2020-07-15:
hi! It would be nice that this is fixed! :)

RussNelson, 2022-09-12:
`M5.Axp.ScreenSwitch(false);` (currently in Github but not pip) will turn off the screen, however it still comes on when it wakes up. I suspect that the default pin settings for the AXP turn the display on. I haven't measured the current it consumes in this state (display turned off), but it's clear that the backlight and LCD are both turned off.

Now to get into the nitty-gritty details. The AXP192 powers the LCD on two pins: the LDO2 and LDO3. Those pins are variable voltage pins. LDO2 powers the backlight, so changing the voltage on that should change the backlight intensity (and hence the heretofore mentioned ScreenBreath. On the other hand, you can turn LDO2 off entirely using `M5.Axp.SetLDO2(false);` . The screen still blinks when it wakes up because the AXP192 gets re-initialized which turns on LDO2 and LDO3. However, it *should* power-down the backlight and LCD.
The default value for both LDO2 and LDO3 is 3.0V, or 0xc. By running ScreenBreath with values between 0 and 100, you can get values between 0x7 and 0xe for the LDO2 value. ScreenSwitch(true) sets it to 0xc and false to 0.
There is no code to change the LDO3 value, likely because the LCD (not the backlight) requires 3.0v.

Note (again) that the current code you get in the Arduino 1.8.13 doesn't have ScreenSwitch and the ScreenBreath parameter must be between 7 and 12, setting the brightness value directly. However, both that version and the version in github have SetLDO2 and SetLDO3, so you can turn them both on or off.

cbira, 2022-09-22:
I cannot test with LDO3 (don't have it in the api I guess). But with LDO2:
Measured using a modified sketch of AXP192 example, battery fully charged (no battery charging current), connected via USB to my laptop, with the code:

loop() { ...
    M5.Axp.SetLDO2(false);                  
    delay(1500);
    Serial.println((int)M5.Axp.GetVBusCurrent());  //<< TP1
    delay(1500);    
    M5.Axp.SetLDO2(true);
    delay(1500); 
    Serial.println((int)M5.Axp.GetVBusCurrent()); // << TP2
    delay(1500); 
}

TP1 returns 37...43 mA
TP2 returns 59...65 mA
About 9...15 mA difference. 

See the below log:
...
01:52:02.096 -> 59
01:52:05.112 -> 43
01:52:08.132 -> 53
01:52:11.148 -> 46
01:52:14.164 -> 59
01:52:17.181 -> 49
01:52:20.163 -> 59
01:52:23.210 -> 37
01:52:26.196 -> 59
01:52:29.214 -> 42
01:52:32.230 -> 65
01:52:35.251 -> 43
01:52:38.236 -> 59
01:52:41.260 -> 49
01:52:44.278 -> 60
01:52:47.294 -> 50
01:52:50.281 -> 55
01:52:53.332 -> 40
01:52:56.315 -> 62
01:52:59.362 -> 45
01:53:02.351 -> 59
01:53:05.365 -> 48
01:53:08.378 -> 62
01:53:11.393 -> 50
01:53:14.407 -> 59
...

cbira, 2022-09-22:
I have updated with the newest git version and with code modified as:
...
    M5.Axp.SetLDO2(false);     
    M5.Axp.SetLDO3(false);             
    delay(1500);
    Serial.println("setLDO2(false), setLDO3(false):" + String((int)M5.Axp.GetVBusCurrent()));
    
    M5.Axp.SetLDO3(true);
    delay(1500);    
    Serial.println("setLDO2(false), setLDO3(true):" + String((int)M5.Axp.GetVBusCurrent()));
    
    delay(1500);        
    M5.Axp.SetLDO2(true);
    Serial.println("setLDO2(true), setLDO3(true):" + String((int)M5.Axp.GetVBusCurrent()));
    
    delay(1500);    
    M5.Axp.SetLDO3(false);        
    Serial.println("setLDO2(true), setLDO3(false):" + String((int)M5.Axp.GetVBusCurrent()));
    
    M5.Axp.SetLDO2(true);     
    M5.Axp.SetLDO3(true); 
    delay(1500); 
...

I got the following values for in mA:

02.11:46.582 -> setLDO2(false), setLDO3(false):51
02:11:49.354 -> setLDO2(false), setLDO3(true):44
02:11:50.846 -> setLDO2(true), setLDO3(true):42
02:11:52.339 -> setLDO2(true), setLDO3(false):66

02:11:55.357 -> setLDO2(false), setLDO3(false):50
02:11:56.881 -> setLDO2(false), setLDO3(true):38
02:11:58.371 -> setLDO2(true), setLDO3(true):44
02:11:59.860 -> setLDO2(true), setLDO3(false):64

02:12:02.906 -> setLDO2(false), setLDO3(false):49
02:12:04.396 -> setLDO2(false), setLDO3(true):50
02:12:05.886 -> setLDO2(true), setLDO3(true):45
02:12:07.409 -> setLDO2(true), setLDO3(false):62

02:12:10.422 -> setLDO2(false), setLDO3(false):51
02:12:11.914 -> setLDO2(false), setLDO3(true):49
02:12:13.438 -> setLDO2(true), setLDO3(true):49
02:12:14.929 -> setLDO2(true), setLDO3(false):69

02:12:17.942 -> setLDO2(false), setLDO3(false):54
02:12:19.467 -> setLDO2(false), setLDO3(true):43
02:12:20.957 -> setLDO2(true), setLDO3(true):43
02:12:22.447 -> setLDO2(true), setLDO3(false):67

02:12:25.495 -> setLDO2(false), setLDO3(false):51
02:12:26.987 -> setLDO2(false), setLDO3(true):45
02:12:28.478 -> setLDO2(true), setLDO3(true):50
02:12:30.001 -> setLDO2(true), setLDO3(false):62