m5stack/M5StickC #25: RTC constructor should not call Wire1.begin()

nopdotcom, 2019-07-29:
In https://github.com/m5stack/M5StickC/blob/3f8f30d6d77c064526723a7d33cd885cbe387ff7/src/RTC.cpp#L5-L7 , `Wire1.begin()` is called in a constructor. This is bad form, and none of the other modules do it. Wire1 setup should be moved into an `rtc.begin()` function.

While it seems to work fine in Arduino, I'm using [PlatformIO](https://platformio.org/) to build the [ENV example](https://github.com/m5stack/M5StickC/tree/master/examples/Hat/ENV). When `Wire1.begin()` is present in the `RTC` initializer, the program crashes *before `setup()`* with an "uninitialized data" diagnostic on serial. Because it crashes before `setup()`, this implies something bad is going on before then — probably in constructors.

georgcampana, 2019-08-17:
@nopdotcom: have just posted a pull request with the fix:  specific begin() method in RTC that initializes the Wire.

EeeeBin, 2019-08-20:
fixed in https://github.com/m5stack/M5StickC/commit/2e8cc6102804f8423fdc95ce28f8c9cfb8e84dea

closed 2019-08-20 by EeeeBin