m5stack/M5StickC #45: RTC interrupt pin connected to ESP32 or not?

huming2207, 2019-08-27:
Hi there,

I would like to know that have you connected the BM8563 RTC chip's INT pin to ESP32 or not? It may be helpful for waking up or notifying the microcontroller.

Regards,
Jackson

EeeeBin, 2019-08-27:
RTC INT pin connect to esp32 GPIO35

closed 2019-08-27 by EeeeBin

huming2207, 2019-08-27:
@EeeeBin Thanks for your reply.

Same question again, what about the gyro's INT pin? Is it connected to ESP32?

EeeeBin, 2019-08-27:
AXP192, IMU and RTC all connect to gpio35
[image]

huming2207, 2019-08-27:
@EeeeBin Thanks. I didn't find it out in the image.

mmarsalko, 2019-10-15:
What about M5Stick variants that use the SH200Q IMU? I've not had luck waking up ESP32 on GPIO35

mmarsalko, 2019-10-16:
@EeeeBin What about M5Stick variants that use the SH200Q IMU? I've not had luck waking up ESP32 on GPIO35 using the SH200's activity interrupts