m5stack/M5StickC #46: LCD controller: ST7735R or ST7735S??

huming2207, 2019-08-27:
Hi there,

Maybe I was wrong but it seems that the screen is ST7735R, same as this one: https://www.adafruit.com/product/3533

...and the datasheet is: https://cdn-shop.adafruit.com/datasheets/ST7735R_V0.2.pdf

Also I suggest you should remove the initialize sequence for other LCD panels: https://github.com/m5stack/M5StickC/blob/master/src/utility/ST7735_Init.h#L12

The VCOM voltages and Gamma LUT are different. If someone wants to port it to other platforms (non-OS or pure ESP-IDF, no Arduino), they may be confused and use the wrong settings. The wrong settings could result in weird color or even killing the panel.

Regards,
Jackson

huming2207, 2019-08-27:
Besides, if you reset the panel by toggling the reset pin (GPIO18), you may not need to call SWRESET again: https://github.com/m5stack/M5StickC/blob/master/src/utility/ST7735_Init.h#L73

I've tested it in my own ESP-IDF BSP library and it works.

huming2207, 2019-08-27:
Furthermore, you don't need to wait 500ms (or 255ms?) after sending SLPOUT command: https://github.com/m5stack/M5StickC/blob/master/src/utility/ST7735_Init.h#L76

On the datasheet, they mentioned:

> When IC is in Sleep In mode, it is necessary to wait 120msec before sending next command because of the stabilization timing for the supply voltages and clock circuits. 

I've tested it with 200ms, it works fine.

closed 2023-07-16 by huming2207

Tinyu-Zhao, 2023-07-17:
The M5StickC LCD controller uses the ST7735SV instead of the ST7735R or ST7735S.