m5stack/M5StickC #66: M5Stick C did not display anything after flash

borbier, 2019-10-05:
I tried flashing my M5StickC with Arduino (just tested hello-world program) The uploading was fine, but the board did not displayed anything and I did not have any evidence for the cause.
I also attached the log here.
~~~
ketch uses 255045 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16652 bytes (5%) of dynamic memory, leaving 311028 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:71:bf:47:4a:94
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 6957.0 kbit/s)...
Hash of data verified.
Compressed 15856 bytes to 10276...

Writing at 0x00001000... (100 %)
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.9 seconds (effective 138.5 kbit/s)...
Hash of data verified.
Compressed 255168 bytes to 130449...

Writing at 0x00010000... (12 %)
Writing at 0x00014000... (25 %)
Writing at 0x00018000... (37 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (62 %)
Writing at 0x00024000... (75 %)
Writing at 0x00028000... (87 %)
Writing at 0x0002c000... (100 %)
Wrote 255168 bytes (130449 compressed) at 0x00010000 in 11.5 seconds (effective 177.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1565.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

~~~

EeeeBin, 2019-10-05:
Try test https://docs.m5stack.com/#/en/core/m5stickc , If  still can't display, may need to contact the seller.

ron2way, 2019-10-06:
I have two mstickc with the mpu6886  that have the same issue with display not working with the factory test.  One arrived with the display operating but stopped working after loading the factory test and one arrived with the display not working but is otherwise functional.  I have 2 msitckc with the sh200q that works with the factory test referenced above and the one included with the m5stickc library.

borbier, 2019-10-06:
> Try test https://docs.m5stack.com/#/en/core/m5stickc , If still can't display, may need to contact the seller.

I tried burning some factory test again and found that it is working. The cause was not the board but from the code that did not support OLED in M5StickC.
(I forgot that the board I used is an OLED version so the LCD functions in the library won't work.)

Thanks anyway

closed 2019-10-06 by borbier

f-rank, 2020-02-18:
I am having the same problem, after arduino ide flashes I get the last line: "Hard resetting via RTS pin..." and nothing happens. Screen is blank and it gets stuck there, though I see through the serial monitor that the program is executing.
@borbier do you happen to remember what you had to change so that the OLED work ?
Thank you

scientistnobee, 2020-06-09:
I have the same issue. I ordered two M5sticks.  One arrived with the display operating but stopped working after loading the factory test and one arrived with the display not working but is otherwise functional. I don't know how to resolve the issue.

EeeeBin, 2020-06-09:
@scientistnobee 
Try follow https://docs.m5stack.com/#/en/core/m5stickc (download firmware by EasyLoader), If still can't display, maybe need to contact the seller or email to support@m5stack.com.

scientistnobee, 2020-06-09:
I tried the EasyLoader but still display not working. Do you know if the screen is connected to the standard SP bus, because I am trying to use the Adafruit lib to see if there is anything that works. So far the LED is working, so I assume it is only the screen that has the problem.

Gitshaoxiang, 2020-06-10:
@scientistnobee just contact the store. use Easyloader the screen still not working , i thought is hardware problem.