m5stack/M5StickC #69: I2c pins override

beckmx, 2019-10-31:
Hello, quick question, I have a library, specifically the CCS811 from adafruit, it is using the default pins, I would like to override the pins or know what i2c pins are really defined for the stick-c

tanakamasayuki, 2019-10-31:
Hi.

https://github.com/espressif/arduino-esp32/blob/master/variants/m5stick_c/pins_arduino.h#L17-L18
static const uint8_t SDA = 32;
static const uint8_t SCL = 33;

My recommendation is to declare and use.
Wire.begin(32, 33);
or
Wire.begin(0, 26);

beckmx, 2019-10-31:
Got it, so it was the grove section, however something seems to happend, I tested under esp8266 and the library works well, in fact they have this lines:

```
Wire.begin();
	#ifdef ESP8266
	Wire.setClockStretchLimit(500);
	#endif
```
I was wondering how to do the same in the m5stick, aparently if I do the same, the compiler doesn't recognize that method

closed 2019-11-24 by beckmx