m5stack/M5StickC #74: LCD: readPixel(), readRect(), readRectRGB() doesn't work

barabanus, 2019-11-11:
ReadPixel(), readRect() and readRectRGB() doesn't return real pixel values from LCD.

For example (for black and white pixels on the screen):
```
for (unsigned y = 0; y < TFT_HEIGHT; ++y) {
 for (unsigned x = 0; x < TFT_WIDTH; ++x) {
   Serial.printf("%s", M5.Lcd.readPixel(x, y) ? "#" : " ");
 }
 Serial.printf("\n");
}
```

This routine prints black pixels though in reality display has white text printed on black background.

EeeeBin, 2019-11-11:
Lcd not support pixel read because spi bus read pixel is very troublesome
 try apply a buffer to flash all the screen，like Sprite https://github.com/m5stack/M5Stack/blob/master/examples/Advanced/Display/Sprite/Sprite_scroll_16bit/Sprite_scroll_16bit.ino 
TFT_eSprite also effective on StickC lib, but sorry， there are currently no examples

barabanus, 2019-11-13:
@EeeeBin Thank you, I did exactly like you said and it worked!

I replaced all my calls to `M5.Lcd` to `LCD::canvas` which has the same interface.
```
namespace LCD {
  TFT_eSprite canvas = TFT_eSprite(&M5.Lcd);

  constexpr uint8_t RIGHT_TO_LEFT = 1;
  constexpr uint8_t LEFT_TO_RIGHT = 3;
  uint8_t orientation = LEFT_TO_RIGHT;

  void setup()
  {
    // set landscape orientation
    canvas.createSprite(TFT_HEIGHT, TFT_WIDTH);
    canvas.setRotation(3);
    M5.Lcd.setRotation(3);
  }

  template <typename T>
  void update(T accel_x)
  {
    uint8_t target = (accel_x > 0) ? RIGHT_TO_LEFT : LEFT_TO_RIGHT;

    if (accel_x != 0 && target != orientation) {
      canvas.setRotation(target);
      M5.Lcd.setRotation(target);
      orientation = target;
    }

    canvas.pushSprite(0, 0);
  }
}
```

closed 2019-11-13 by EeeeBin