m5stack/M5StickC #77: MacOS catalina upload problem

beckmx, 2019-11-24:
Hello, I have been having problems with the latest macos version, looks like it cannot longer communicate with the module in order to upload sketches from arduino or via the esptool, is there a work around for this?

EeeeBin, 2019-11-24:
Could you show upload failed log
make sure baudrate is 1500000 or 750000 or 115200

beckmx, 2019-11-24:
> sptool.py v2.6
> Serial port /dev/cu.usbserial-9D52BE9945
> Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
> _
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
>

EeeeBin, 2019-11-24:
First, make sure uart port select M5StackC
Then make sure select Tools->Board->M5StickC , Tools -> Ports(choose the series port name) Upload Speed->115200,  then upload
or 
test https://docs.m5stack.com/#/en/quick_start/m5stickc/m5stickc_quick_start_with_uiflow
If all failed, please contact the seller

beckmx, 2019-11-24:
Yes, still failed, this only happens in MacOS Catalina 10.15, I changed the speed and still cannot be uploaded

EeeeBin, 2019-11-24:
Sorry, None of our Mac are the latest system, so I can't test what happens in the latest MacOS

tanakamasayuki, 2019-11-25:
Hi!

https://qiita.com/SamAkada/items/fa32d6072a832a10cd84

https://github.com/espressif/arduino-esp32/issues/2278
GPIO0 connected to GND(DOWNLOAD_BOOT)

or

Installing beta version 1.9.0 of Arduino IDE.

masrodjie, 2020-07-01:
Please check this https://forum.m5stack.com/topic/1591/m5stickc-atom-on-macos-catalina-can-t-upload-esp32-timed-out-waiting-for-packet-header-solution
It's works on me

Zontex, 2020-11-12:
Hello, we confirmed that it works well for OSX Catalina, make sure you have the latest version of the driver, you can take a look also at the forum post above, could be helpful.

closed 2020-11-12 by Zontex