m5stack/M5StickC #78: How to put LCD to sleep

takanotume24, 2019-12-06:
I want to save power during DeepSleep, so I want to make the LCD sleep. I found “Sleep-In Mode” on page 30 of https://github.com/m5stack/M5-Schematic/blob/master/Core/ST7735S_v1.1.pdf. Can I migrate from software to this?
Or is there a way to save power as well?
I tried M5.Axp.ScreenBreath (7), but it seems that the battery runs out within one day even with DeepSleep.

EeeeBin, 2019-12-06:
Try test it: 
https://github.com/m5stack/M5StickC/blob/master/examples/Advanced/AXP192/sleep/sleep.ino
**_M5.Axp.DeepSleep(SLEEP_SEC(20))_** will close tft 
By test DeepSleep also used **_2~3 ma_** current

closed 2019-12-06 by EeeeBin

dave-sausages, 2020-06-28:
I've found an issue: If you call "M5.Axp.DeepSleep" but then press the power button once, the current consumption goes from 3ma to 17ma but doesn't wake up. It took me hours to work out why my battery kept going flat. How do I avoid this?

tanakamasayuki, 2020-06-28:
AXP sleep is not the same as ESP32 sleep.
The power button clears AXP sleep.
They are not interlocked.
Must be checked by INT interrupt or polling from ESP32.

I don't recommend either...