m5stack/M5StickC #80: Yaw value keeps increasing/decresing even without movement

jokandre, 2019-12-22:
Hi there, 
I have been trying to debug getAhrsData(&pitch,&roll,&yaw) without success.
With the M5StickC on the table without movement the values I get for yaw keeps growing in circles.

I did take a look into MahonyARSupdate and other versions available on the internet but still cant get it to work properly.

This happens only for yaw. Pitch and  roll values are quite stable after few seconds.

Gist of the values:
<script src="https://gist.github.com/jokandre/be502bc3ed397fcd13013df5cd740ddd.js"></script>

jokandre, 2019-12-22:
Any help is welcomed since I couldn't find much information about the mpu6886 chip. Thanks in advance

EeeeBin, 2019-12-22:
mpu6886 datasheet in here: https://github.com/m5stack/M5-Schematic/blob/master/datasheet/MPU-6886-000193%2Bv1.1_GHIC.PDF.pdf  Mostly the same as _**mpu6050**_
Gyro yaw drift is normal, if  want to get stable yaw angle, it need to use the magnetometer calibrate

closed 2019-12-22 by EeeeBin

jokandre, 2019-12-22:
Any advice on how to get the magnetometer calibrated?