m5stack/M5StickC #81: Using M5.6886 api, yaw will always drift

jokandre, 2019-12-22:
Hi there, 
I have been trying to debug getAhrsData(&pitch,&roll,&yaw) without success.
With the M5StickC on the table without movement the values I get for yaw keeps growing in circles.

I did take a look into MahonyARSupdateIMU and other versions available on the internet but still cant get it to work properly.

Gist of the values:
<script src="https://gist.github.com/jokandre/be502bc3ed397fcd13013df5cd740ddd.js"></script>

**Advice on how can I fix it is much needed.** 
I know that it needs to calibrate. But there is no documentation about it. Also there is nothing on the code inside the library to do it.
If I should not use the M5StickC library at least tell me what should I use instead. Thanks

EeeeBin, 2019-12-22:
If only use mpu6886, it is impossible to obtain a stable yaw value
Calibrate is mean use other chips get yaw angle value, such as magnetometer chip
if want to know detail or why, pls try use _**google**_ search for this question

jokandre, 2019-12-22:
@EeeeBin Actually all I want is to map data from mpu6886 to a 3D object. It does not have to be with yaw. I have seen it being done with mpu6050 and since they are similar I thought it was possible. Am I wrong?

EeeeBin, 2019-12-22:
try this example  https://github.com/m5stack/M5StickC/blob/master/examples/Basics/FactoryTest/FactoryTest.ino
There is a 3d cube related by mpu6886, but it is not perfect, you can refer to this
And mpu6886 same as mpu6050 If not use dmp, you can use mpu6050 lib in mpu6886

jokandre, 2019-12-22:
I updated the M5StickC library that was on my ArduinoIDE (it was missing a lot of newer examples).
To run the example you recommended I installed FastLED by Daniel Garcia.
But compiling on ArduinoIDE says:
```
Sketch uses 1372291 bytes (104%) of program storage space. Maximum is 1310720 bytes.
Global variables use 52584 bytes (16%) of dynamic memory, leaving 275096 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

Error compiling for board M5Stick-C.
```

EeeeBin, 2019-12-22:
board select page choose partition scheme -> large app

jokandre, 2019-12-22:
@EeeeBin Thank you very much!!!
The example is working now. I will study it.

closed 2019-12-22 by jokandre

amcchord, 2020-06-11:
While perhaps no the most sophisticated fix, a simple calibration of the gyro makes a wold of a different in improving yaw performance when the MPU6886 is still. I added the ability to calibrate the gyro to the MPU6886 library. It appears 2 or 3 seconds at startup while the device is still can make the yaw stable when idle.