m5stack/M5StickC #82: AXP192 battery current

karfas, 2019-12-29:
A few days ago I ported the Arduino AXP192 code to micropython (see https://github.com/karfas/M5StickC-uPy). 
It works, but some questions arised:

1) Has someone a pointer to an **English** datasheet of the AXP192 ? 

2) In AXP192::GetBatCurrent(), the current to the battery is calculated from 13 bits read from the AXP address 0x7a.
In AXP192::GetBatChargeCurrent(), the battery current is read from 12 bits at address 0x7a.
Code is pasted below.

Which version is correct ? The ReadXXBit() functions are also not helpful - these shift the first byte read 3/4 bits and ADD the second byte without any mask).

Kind regards,

Thomas

```C
float AXP192::GetBatCurrent()
{
    float ADCLSB = 0.5;
    uint16_t CurrentIn = Read13Bit( 0x7A );
    uint16_t CurrentOut = Read13Bit( 0x7C );
    return ( CurrentIn - CurrentOut ) * ADCLSB;
}
```
```C
float AXP192::GetBatChargeCurrent()
{
    float ADCLSB = 0.5;
    uint16_t ReData = Read12Bit( 0x7A );
    return ReData * ADCLSB;
}
```

EeeeBin, 2019-12-30:
According to the AXP documentation
read **13 bits** from the AXP address 0x7a is correct, same as 0x7c

closed 2019-12-30 by EeeeBin

karfas, 2019-12-30:
In this case, someone should fix AXP192::GetBatChargeCurrent() **before** closing this issue.