m5stack/M5StickC #83: TFT_eSprite drawString is crash

nnn112358, 2020-01-05:
Kernel panic occurs when the size of drawString is set to 1 in TFT_eSprite.
```
moji2 = "Test";
TFT_eSprite * Spr->drawString(moji2, 80, 0, 1);
```

` Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`

There is no problem if the character size is 2 or more.
```
moji2 = "Test";
TFT_eSprite * Spr->drawString(moji2, 80, 0, 2);
```

I use
・arduino-esp ver 1.04
・m5stickc libs in adrduino ver 0.1.1

EeeeBin, 2020-01-05:
Hi.
Try follow:  https://github.com/m5stack/M5Stack/tree/master/examples/Advanced/Display/Sprite

nnn112358, 2020-01-06:
This example is no problem by the character size is 2 or more.

```
EX.
    stext2.drawString("Hello World", 6, 0, 2); // draw at 6,0 in sprite, font 2 size
```

If floow modification is bad.

```
    stext2.drawString("Hello World", 6, 0, 1); // draw at 6,0 in sprite, font 1 size
```

nnn112358, 2020-01-07:
I investigated.

**int16_t TFT_eSPI::drawString(const char *string, int32_t poX, int32_t poY, uint8_t font)**

When specifying the "uint8_t font" value, there is no problem between 2 and 8.
"uint8_t font" value is 1 or over 9will crash.

My Trouble Source Code

```
  //This is crash
  //Spr->drawString(moji, 0, 0, 1);

  //This is No problem
  Spr->drawString(moji, 0, 20, 2);
  Spr->drawString(moji, 0, 40, 3);
  Spr->drawString(moji, 0, 60, 4);
  
  //This is crash
  //Spr->drawString(moji, 0, 60, 10);
```
https://gist.github.com/anoken/8ef9355e6945080b37fed12d6edb1e94

**Crash Point**
drawString() goes into textWidth() processing

TFT_eSPI::textWidth(const char *string, uint8_t font)
https://github.com/m5stack/M5StickC/blob/master/src/utility/In_eSPI.cpp#L2354

If  "uint8_t font" value between 2 and 8 is follow process.
int16_t TFT_eSPI::textWidth(const char *string, uint8_t font)
https://github.com/m5stack/M5StickC/blob/master/src/utility/In_eSPI.cpp#L2387

But, If  "uint8_t font" value1 or 9 is follow process.
gfxFont value is crashed when accessed because gfxFontis NULL.
https://github.com/m5stack/M5StickC/blob/master/src/utility/In_eSPI.cpp#L2409

ciniml, 2020-01-07:
Hi,

I found that the root cause of this issue is just initialization code of TFT_eSPI::gfxFont is missing in its constructor.

Adding `gfxFont = NULL;` before the line 269 of `In_eSPI.cpp` is enough to solve this issue.

mancini0, 2020-07-20:
> Hi,
> 
> I found that the root cause of this issue is just initialization code of `TFT_eSPI::gfxFont` is missing in its constructor.
> 
> Adding `gfxFont = NULL;` before the line 269 of `In_eSPI.cpp` is enough to solve this issue.

@cinimi's workaround worked for me - thanks!