m5stack/M5StickC #85: M5StickC CardKB First Time usage

markbirss, 2020-01-10:
Hi

I run the CardBK example with the CardBK connected but get no repsonse on screen or when press keys.

Do I need to flash the CardBK first with the keyboard firmware or should it be already loaded?

pierreverbakel, 2020-01-18:
Hello Mark,

I had also a non functioning test-app with the CardKB which I received today for M5StickC. I guessed it might have something to do with the Pin Number, since it is probably a ported App from M5Stack, that does not have  HAT-Connector.
So I changed the Wire.begin() line to Wire.begin(0, 26, 400000); 
Also I commented out the lines referring to GPIO5, since it would not work with them in. 
Maybe you get it working with the same change.

`#include <M5StickC.h>
#include <Wire.h>

#define CARDKB_ADDR 0x5F

void setup(){
  M5.begin();
  Serial.begin(115200);
  //Wire.begin();
  Wire.begin(0, 26, 400000); // Changed by PV
  //pinMode(5, INPUT); // PV: What is on Pin 5?
  //digitalWrite(5, HIGH);
  M5.Lcd.fillScreen(BLACK);
  M5.Lcd.setRotation(3);
  M5.Lcd.setCursor(1, 10);
  M5.Lcd.setTextColor(YELLOW);
  M5.Lcd.setTextSize(2);
  M5.Lcd.printf("IIC Address: 0x5F\n");
  M5.Lcd.printf(">>");
}

void loop(){
  Wire.requestFrom(CARDKB_ADDR, 1);
  while(Wire.available())
  {
    char c = Wire.read(); // receive a byte as characterif
    if (c != 0)
    {
      M5.Lcd.printf("%c", c);
      Serial.println(c, HEX);
      //M5.Speaker.beep();
    }
  }
  // delay(10);
}`

markbirss, 2020-01-18:
@pierreverbakel 

I will get back to you once I test you suggested changes, thank you

markbirss, 2020-01-19:
@pierreverbakel 

YES !

your suggested changes in fact solve both the issues i had, thank you !

ie

1. The I2C_Tester M5StickC example would not detect my M5Stack CardBK on 0x5F, this is solved changing the example to specifically setting the connected pins 

`  Wire.begin(32, 33, 400000);`

[image]

[image]

and

2. The M5StickC CardBK example working with your suggested removal of 

```
  //pinMode(5, INPUT);
  //digitalWrite(5, HIGH);
```

and leaving 

`  Wire.begin();`

for me since I have the CardBK connected to the Grove port (GND, 5V, G32,G33)

[image]

also I re-flashed my other CardKB with another hex file to support PS/2 mode then using a PS2-to-USB adapter the be used as a USB keyboard (Japanese layout)

[image]

markbirss, 2020-01-19:
I notice even without setting 

`  Wire.begin();`

also works now

pierreverbakel, 2020-01-19:
Hey Mark,
Great ilustrated update on your work! I did not realize you have the standard keyboard, since I got myself a M5StickC-version, that did not work at first. 
So now both are supported, at least when using the correct Wire.begin(x,y,z);
M5.begin() does a lot of initialisation. I did not dive into that, but most probably it has the right Wire.begin(x,y,z)-call in it already for the Grove-Connector.  

- Nice GPIO-Layout, it will come in handy. 
- Nice cable you got, where would one get such a cable.
- Nice keyboard cover you have made, love it.

markbirss, 2020-01-19:
@pierreverbakel 

Ah ok,

cable made up myself, with another firmware, cut open PS2toUSB converter and connected USB side to CardKB

I used Raspberry Pi to flash keybaord with new hex file

https://github.com/markbirss/CardKeyBoard_PS2

This is the keyboard cover
https://www.thingiverse.com/thing:3544822

Zontex, 2020-11-12:
Hello, is the issue solved? if not please let us know how can we help. I'm closing this issue for now, if it's not solved - feel free to re-open it.

closed 2020-11-12 by Zontex