m5stack/M5StickC #86: MacOs trouble with connection

maxsimak, 2020-01-14:
There are some problems when flashing Arduino projects (a fatal error occurred: failed to connect to ESP32: invalid packet header (0x4F)) Download speed and board are correct. When I flash in Ui Flow the exact same problem. Drivers set from quick guide. How fix this?
[image]

EeeeBin, 2020-01-15:
Maybe same as https://github.com/m5stack/M5StickC/issues/77
Detail please contact after-sales of M5

maxsimak, 2020-01-15:
Maybe there is some other solution. I would not like to make a refund

maxsimak, 2020-01-15:
M5 StickC works correctly, but the problem with the drivers on the Mac .If you have friends with Windows 10 PC or BootCamp (work without drivers) then flash in M5 Burner Ui Flow and use it in Cloud mode

closed 2020-01-15 by maxsimak

masrodjie, 2020-07-01:
Please check this https://forum.m5stack.com/topic/1591/m5stickc-atom-on-macos-catalina-can-t-upload-esp32-timed-out-waiting-for-packet-header-solution
It's works on me