m5stack/M5StickC #100: Is it possible to read the IR transmitter (or LED) to detect ambient light?

PaulMcGuinness, 2020-03-27:
I've seen various hacks on Hackaday.com to use LED's as sensors.

Does anyone think this could be leveraged within the M5StickC?

danipixis, 2020-05-21:
Looking at the datasheet I can see the IR should be readable. But IR (infrared) are not in the field of visible so I guess the most you might dewtect is the presence of some IR rays and not of light.

dcremonini, 2020-05-21:
Looking at the datasheet I can see the IR should be readable. But IR (infrared) are not in the field of visible so I guess the most you might dewtect is the presence of some IR rays and not of light.

PaulMcGuinness, 2020-05-21:
Agreed, but there is a lot more IR in daylight that at night, so ergo you could correlate a value I would have thought?

dcremonini, 2020-05-22:
I don't know and I received the device a few days ago. I will keep in mind your question and I hope to having the change to have a look.

Zontex, 2020-11-16:
not possible, for this you have other type of sensor such as photo-resistor or photo-diode that convert light into electrical energy, IR can be used as proximity sensors (read the light hit on object and bounce back time) but not as photo-sensors. also, consider asking such questions at forum.m5stack.com you'd probably get more answers from the rest of the community members, this git is more of M5Stick related issues :)

closed 2020-11-16 by Zontex