m5stack/M5StickC #103: In_espi min() conflict with sstream

happyme531, 2020-05-16:
compile simple code like this:
```
#include <M5StickC.h>
#include <iostream>
#include <sstream>

void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
and get error:
```
In file included from C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/M5Display.h:7:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/M5StickC.h:70,

                 from C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_429256\BareMinimum.ino:1:

c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\streambuf.tcc: In member function 'virtual std::streamsize std::basic_streambuf<_CharT, _Traits>::xsgetn(std::basic_streambuf<_CharT, _Traits>::char_type*, std::streamsize)':

C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token

   #define min(a,b) (((a) < (b)) ? (a) : (b))

                    ^

c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\streambuf.tcc: In member function 'virtual std::streamsize std::basic_streambuf<_CharT, _Traits>::xsputn(const char_type*, std::streamsize)':

C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token

   #define min(a,b) (((a) < (b)) ? (a) : (b))

                    ^

c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\istream.tcc: In member function 'std::streamsize std::basic_istream<_CharT, _Traits>::readsome(std::basic_istream<_CharT, _Traits>::char_type*, std::streamsize)':

C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token

   #define min(a,b) (((a) < (b)) ? (a) : (b))

                    ^

c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\sstream.tcc: In member function 'virtual std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::overflow(std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type)':

C:\Users\Administrator\Documents\Arduino\libraries\M5StickC\src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token

   #define min(a,b) (((a) < (b)) ? (a) : (b))

                    ^
```
comment the code in In_espi.h, and compiling passes.
```

#ifndef min
  // Return minimum of two numbers, may already be defined
  #define min(a,b) (((a) < (b)) ? (a) : (b))
#endif
```

Zontex, 2020-11-17:
Hello, seems like you figured out the compilation issue by work-around. we don't officially support sstream yet but we will look into it to possibly give better compatibility between the sstream and our framework in the future.

closed 2020-11-17 by Zontex

error23, 2021-05-26:
i have the same isssue with the #include <queue>
:'(