m5stack/M5StickC #108: arduino 1.8.12 & FactoryTest produces a sketch bigger than flash

RussNelson, 2020-05-24:
Downloaded a fresh 1.8.12.
Downloaded the board and libraries
-     M5Stack 0.3.0
-     M5Stick 0.2.0
-     FastLED 3.3.3

Loaded examples/Basics/FactoryTest
Tried to compile it with no changes.
Got this error:
```text section exceeds available space in board
Sketch uses 1348331 bytes (102%) of program storage space. Maximum is 1310720 bytes.
Global variables use 52664 bytes (16%) of dynamic memory, leaving 275016 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board M5Stick-C.
```

EeeeBin, 2020-05-24:
Try to read note
https://github.com/m5stack/M5StickC/blob/0.2.0/examples/Basics/FactoryTest/FactoryTest.ino#L5

RussNelson, 2020-05-24:
Well, what do you know!?! That fixed it. Thank you for helping me with my reading skills. :-)

closed 2020-05-24 by RussNelson