m5stack/M5StickC #110: Unable to receive interrupts from AXP192

fustilio, 2020-05-29:
I've been trying to receive interrupts from the AXP192 via GPIO35 to (hopefully) detect the power button press as an interrupt. Button A and B have no issues with this as they are directly hooked up to GPIO37 and GPIO39 respectively,
Here are some code snippets:
```
// Method in AXP192.cpp to enable all IRQ flags by writing to registers 0x40, 0x41, 0x42, 0x43
void AXP192::enableAllIRQ() {
    uint8_t val = 0xFF;
    for (int i = 0; i < 4; i++) {
        Write1Byte(0x40 + i, val);
    }
}

// method to be called when interrupt is triggered
void IRAM_ATTR isr() {
    axpIrq = 1;
}

// The following methods are called during setup
void setup() 
{
// other functions...
// setup GPIO35 as input
pinMode(35, INPUT_PULLUP);
// attach interrupt service routine
attachInterrupt(35, isr, FALLING);
// some method to clear IRQs
}

void loop()
{
    // trigger if axpIrq flag is set
    if (axpIrq) {
        Serial.printf("INTERRUPT DETECTED\n");
        axpIrq = 0;
    }
}
```

That's the rough gist of the code I'm working with now. Is there anything else I'm missing?

tanakamasayuki, 2020-05-29:
The GPIO35 cannot be pulled up for input only.
Someone other than GPIO35 will have to do the pull up.
However, the circuit has not been pulled up.

The MPU6886 can be pulled up.
Try running IMU.Init().
Please note that this is not a formal method.

EeeeBin, 2020-05-29:
First: Make imu and rtc int pin value to 1
Then: Clear AXP192 IRQ status
Then: you can read axp192 irq

fustilio, 2020-05-29:
Thanks for the tips @tanakamasayuki @EeeeBin I'll give them a shot! :-)
I'll report here/make a PR if I have any findings.

EeeeBin, 2020-05-29:
here is example
```
#include "M5StickC.h"

static void axp192_disableAllIRQ() {
  M5.I2C.writeByte(0x34, 0x40, 0x00);
  M5.I2C.writeByte(0x34, 0x41, 0x00);
  M5.I2C.writeByte(0x34, 0x42, 0x00);
  M5.I2C.writeByte(0x34, 0x43, 0x00);
}

static void axp192_clearAllIRQStatus() {
  M5.I2C.writeByte(0x34, 0x44, 0xff);
  M5.I2C.writeByte(0x34, 0x45, 0xff);
  M5.I2C.writeByte(0x34, 0x46, 0xff);
  M5.I2C.writeByte(0x34, 0x47, 0xff);
  M5.I2C.writeByte(0x34, 0x4d, 0xff);
}

static void axp192_enableBtnIRQ() {
  M5.I2C.writeByte(0x34, 0x42, 0x03);
}

static uint8_t axp192_getBtnIRQStatus() {
  uint8_t status;
  M5.I2C.readByte(0x34, 0x46, &status);
  return status;
}

void setup() {
  M5.begin();
  Wire.begin(21, 22, 400000);
  pinMode(35, INPUT);
  axp192_disableAllIRQ();
  axp192_clearAllIRQStatus();
  axp192_enableBtnIRQ();
}

void loop() {
  delay(1000);
  Serial.printf("axp192 btn status: %d\r\n", axp192_getBtnIRQStatus());
  Serial.printf("pin 35 status: %d\r\n", digitalRead(35));
}
```

fustilio, 2020-05-29:
@EeeeBin I've tried out the example, the digitalRead of pin35 is always 0. 

I gave the problem another in-depth look. 😂😂

[image]
The above segment from datasheet seems to indicate two modes of operation "方式 A" and "方式 B" .  It seems the IRQ pin doesn't do anything in  "方式 B" which was why I investigated further. The following two segments indicate that if LDO1 and SYSEN are connected, AXP192 is in "方式 A" otherwise it is in "方式 B". 

[image]
[image]

Combining this with the following excerpt from the schematic. If both LDO1 and SYSEN are connected, then it follows that by default  "方式 A" but when I read the mode from register 0x01, it seems the mode is in "方式 B" which is puzzling.

[image]

Below is a snippet of the datasheet regarding reading from register 0x01 
[image]

For now, I'm checking to see if I'm reading from register 0x01 (that indicates the 方式) correctly and hoping someone from the team can shed some light/join me in my quest for the interrupt.

EeeeBin, 2020-06-01:
@fustilio
It need set imu  Interrupt active value, default active value is high, so pin35 is always low
fixed example in https://github.com/m5stack/M5StickC/commit/df5c99f4cea87e5354d2b63f22eb937ffd90a0f3

fustilio, 2020-06-01:
Thanks @EeeeBin for the example provided. The GPIO pin seems to be working as expected, I should (hopefully) be able to modify it for my needs. Also I've made a PR https://github.com/m5stack/M5StickC/pull/111 that should fix some issues with that example.

EeeeBin, 2020-06-01:
Merged https://github.com/m5stack/M5StickC/pull/111

closed 2020-06-01 by EeeeBin