m5stack/M5StickC #113: Color defines causes compiler errors

corneliusmunz, 2020-06-03:
The color defines in https://github.com/m5stack/M5StickC/blob/1befa455553c965aeb7c91e8515a9f192310f7b4/src/utility/ST7735_Defines.h#L94-L112
causes problems when compiling with other libraries or other code which uses the commonly names for colors as Variable or Enum names.  The namespaced color defines in are much better and should be used only
https://github.com/m5stack/M5StickC/blob/1befa455553c965aeb7c91e8515a9f192310f7b4/src/utility/ST7735_Defines.h#L74-L92

**Can you please delete the common color name defines in https://github.com/m5stack/M5StickC/blob/master/src/utility/ST7735_Defines.h to prevent compiler errors with other code?** 

In my case i have a local enum definition of color in a class of the following library and here the Enum names are replaced by the Macro definition of your library:
https://github.com/corneliusmunz/legoino/blob/edc40fa058e640bf455411be0708832a0e13b093/src/Lpf2Hub.h#L75-L89
This replacement leads to error messages like the following
```
In file included from /home/programmez/Arduino/libraries/M5StickC/src/utility/In_eSPI_Setup.h:17:0,
                 from /home/programmez/Arduino/libraries/M5StickC/src/utility/In_eSPI.h:24,
                 from /home/programmez/Arduino/libraries/M5StickC/src/M5Display.h:7,
                 from /home/programmez/Arduino/libraries/M5StickC/src/M5StickC.h:70,
                 from /tmp/arduino_modified_sketch_117376/sketch_jun02a.ino:1:
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected identifier before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected '}' before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected unqualified-id before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
In file included from /home/programmez/Arduino/libraries/Legoino/src/PoweredUpHub.h:14:0,
                 from /tmp/arduino_modified_sketch_117376/sketch_jun02a.ino:2:
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:89:1: error: expected declaration before '}' token
 };
 ^
exit status 1
Error compiling for board M5Stick-C.
```

EeeeBin, 2020-06-03:
Eidt this will cause some example or user code error 

Maybe edit your code 
or 
used `#undef` will be better

corneliusmunz, 2020-06-03:
Hi @EeeeBin! Thx for your fast response! 
I can also change it in my library (https://github.com/corneliusmunz/legoino) but i am not happy about that because i have a enum definition inside a propper class and that names are hard replaced by the Macro of the m5stickc library.  For the future it would be better to use a constant instead of a macro. The constant will not replace any "BLACK" or "RED" in the overall code by a value and is as efficient as a macro 
`const uint16_t BLACK = 0x0000;`

If i use #undef it is strongly dependent on the order of the header includes. Maybe you can think of replacing the macro defines with const definitions. This should be transparent for all your users and the example code