m5stack/M5StickC #120: The IMU (MPU6886) does not work properly when running on a different core using FreeRTOS

meltingrabbit, 2020-07-16:
When changing the execution core of the task to read IMU (MPU6886) values using `xTaskCreatePinnedToCore` from 1 to 0, the correct value cannot be read from the IMU.

In the sample code below, if you swap lines 20 and 21, it works or doesn't work properly.

```
#include <M5StickC.h>
#include <esp_task.h>

void multitask_imu(void* arg) {
  static int16_t x,y,z;
  while (1) {
    M5.IMU.getAccelAdc(&x,&y,&z);
    Serial.printf ("%d,%d,%d\n", x, y, z);
  }
}

void setup() {
  M5.begin();
  M5.IMU.Init();
  Serial.begin(115200);

  disableCore0WDT();
  disableCore1WDT();

  xTaskCreatePinnedToCore(multitask_imu, "Task0", 512*4, NULL, ESP_TASK_PRIO_MAX, NULL, 0);
  // xTaskCreatePinnedToCore(multitask_imu, "Task0", 512*4, NULL, ESP_TASK_PRIO_MAX, NULL, 1);

  delay(1000);
}

void loop() {
  delay(10000);
}
```

EeeeBin, 2020-07-16:
Hi, meltingrabbit
Arduino run loop() in core1, so if use core 1 to run task, it will run normal
If use core0, I can't be sure i2c is working well (now i2c running looks weird)

meltingrabbit, 2020-07-16:
Thank you for your kind replay.
I got it.

Do you know how to change the working core of the I2C tasks?

geiseri, 2020-11-22:
They use global variables and are hardcoded in a few places.  Your best bet is to replace their i2c calls in `MPU6886::I2C_Read_NBytes` and `MPU6886::I2C_Write_NBytes` with the platform i2c calls.  Note you will also need to replace the `Wire1.begin(21,22);` in  `MPU6886.cpp` with the correct i2c setup code.  In general, unless you are doing a very simple project you will need to replace all of their hardware calls with platform code.  Also keep an eye out for random hardcoded values.  Lastly check the other projects for MPU6886.h/cpp files.  They copy,  paste, and modify slightly in all of the projects, so some may or may not have fixes or changes depending on when/if they were updated.