m5stack/M5StickC #124: Need IMU Offset calibration

Menginventor, 2020-11-19:
I use MPU6886 and I need to read/write acc/gyro offset value for calibrating IMU.
please add these feature to your library.
regard

uecken, 2020-11-22:
Also, I have a favor.
After execute function "M5.IMU.getAhrsData(&pitch,&roll,&yaw)", I can get about pitch:3, roll:10, yaw:random.
I want to offset this pitch and roll within -180 to 180 phase.
Could you fix it?

Menginventor, 2020-11-23:
May be you need to implement AHRS update function by yourself.
the AHRS (roll pitch yaw) system is define how imu frame rotate from global frame you can redefine imu frame by tranform them once with rotation matrix before assign to AHRS function.

uecken, 2020-11-23:
Thank you your comment. I'm not used to C++, I'll try to fix.

edumo, 2021-05-03:
9 axis working example https://github.com/edumo/M5StickC_Imu-9-axis