m5stack/M5StickC #125: Mahony AHRS implement in IMU example is wrong.

Menginventor, 2020-11-19:
In MahonyAHRS.cpp specify update frequency to 25Hz but the example use delay of 1000 ms.
So the algorith is wrong and this code is useless.

geiseri, 2020-11-22:
Ignore their copy, it uses hardcoded and global variables that are for the most part wrong.  Just use the https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/ version and save yourself tons of heartache.   You can replace their `MPU6886::getAhrsData`.  Also, note that samples will be a bit off unless you can ensure that the average sample rate is in fact what you put in the `Mahony::begin(...)`.  In my case, I ended up getting the average time for the "loop" call for a few seconds and then using that for the sampling frequency.   Sadly you will find the numbers are pretty much garbage though due to #124.

Menginventor, 2020-11-23:
Hmm what the product....  I really hope some bugfix  from the source.

geiseri, 2020-11-25:
Oh, these guys have sweet hardware, but their code is a trainwreck :)  I am actually working on consolidating them all right now.  The StickC is awesome for ESPHome.