m5stack/M5StickC #126: M5Display.h declares a setBrightness but the method is not implemented

IsaacG, 2020-11-29:
https://github.com/m5stack/M5StickC/blob/master/src/M5Display.h#L34

There's a M5Display:setBrightness in the .h but not the .cpp

Is there supposed to be an implementation?

ldoyle, 2020-11-30:
Indeed the function implementation seems missing. Unfortunately, it will not be easy to set an arbitrary brightness on M5StickC.
According to the schematics, the backlight LED is connected directly to LDO2 of the AXP power controller.
Inside AXP192.h/.cpp the function `ScreenBreath` does something to try and set the brightness via the LDO voltage.
Due to the limit of `12`, it effectively tunes the LDO voltage from 1.8V to 3.0V which will result in some brightness change, for sure.
You could try and simply implement a call of AXP192's `ScreenBreath` inside `M5Display::setBrightness` and create a Pull Request?

Tinyu-Zhao, 2021-09-02:
Using M5.Axp.SetLcdVoltage() is a correct choice

Tinyu-Zhao, 2021-09-02:
Using Axp to manage power related is the right choice, I am also on the minimum voltage

closed 2021-09-02 by Tinyu-Zhao