m5stack/M5StickC #135: Factory Demo no longer fits in M5StickC

seclorum, 2021-06-22:
Hi,

As reported here [1], it seems the Factory demo no longer fits in the stock M5StickC (i.e. the one that comes with the Bala-C kit) .. I tried to restore the Factory demo after playing with the Bala-C code, but I get:

```

Sketch uses 1374114 bytes (104%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 47148 bytes (14%) of dynamic memory, leaving 280532 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board M5Stick-C.

```

[1] - https://community.m5stack.com/topic/2257/m5stickc-restore-factory-demo-firmware/3

Tinyu-Zhao, 2021-08-16:
You can try modifying the Settings as shown in the following image or use Platform IO to compile the program
[image]

seclorum, 2021-08-18:
Okay thanks for the info - although I think this should be updated in the code somehow (maybe with a switch) - if you re-open the issue, I will submit a PR this week to fix it for future devs who want to learn from the onboard-factory demo.

clintonefisher, 2021-12-24:
> Okay thanks for the info - although I think this should be updated in the code somehow (maybe with a switch) - if you re-open the issue, I will submit a PR this week to fix it for future devs who want to learn from the onboard-factory demo.
> 
> EDIT: btw, does this mean there is an M5StickC hardware configuration with much more RAM?

Hi Seclorum, perhaps my troubles can help you.    I purchased a M5StickCPlus and was using the M5StickC library to build the FactoryTest.   I was also running into the space issue.   I tried using board_build.partitions = no_ota.csv in my platformio.ini file which expands the program memory into the file-system space.  This got it to build, but then the display wouldn't work properly.   

Turns out, I was using the wrong library!!   So I started fresh, and used the M5StickCPlus library, and then built the FactoryTest.ino app that was included with the M5StickCPlus library, and it worked perfectly just like the original, fitting into the available memory with a couple-hundred K to spare.

So maybe check to make sure that you're using the library and example code for your exact M5StickC board model.
Hoping this helps!!

seclorum, 2021-12-24:
Nice, thanks for the tip - I have since moved on to using my m5stickC for other things but I’ll check out the factory demo again soon since  this is a very interesting  way to learn more about the layout…

;
--
seclorum

closed 2023-01-30 by Tinyu-Zhao