m5stack/M5StickC #155: Suggestion for M5StickC TOF example

MDAR, 2022-05-10:
Thanks for all your hard work.

Can I suggest a small tweak to the example code for the ToF ?

I didn't know there are TWO TOF sensors ava;able, so I took this example as being the ONLY working option.,

https://github.com/m5stack/M5StickC/blob/master/examples/Hat/TOF/TOF.ino

When I discovered that I had the Grove version and NOT the HAT version, I realised that I needed to change the I2C ports.

Can I suggest that `line 14` be changed to

``` 
    Wire.begin(32, 33, 100000UL); // (32, 33, 100000UL) for Grove socket - (0, 26, 100000UL) for HAT I2C
```

Just to save someone else having to make the same discovery.

I can report that the example works PERFECTLY now that I have made this change

Many Thanks
Stuart

HAT Version
[image]

Vs
Grove version
[image]

Tinyu-Zhao, 2022-05-30:
Thank you for your feedback. I will add this prompt to the next version.

closed 2022-05-30 by Tinyu-Zhao

MDAR, 2022-05-30:
Thanks for noticing my comment.

I should have mentioned that I ran into a little issue when I ran the code, after I changed the I2C ports.

I would seem that the Grove device wasn't ready when the m5StickC (original or Plus) polled it in the setup void.

This version of the code works, because I moved the `Wire.begint` a bit further into the script.

```
// please install vl53l0x lib first (https://github.com/pololu/vl53l0x-arduino)
// lib in Sketch->Includ Library->Library Manager, search for vl53l0x, Author:
// pololu

#include <VL53L0X.h>
#include <Wire.h>

  #include "M5StickC.h"
//      #include <M5StickCPlus.h>

String Message = "" ;
int distanceTreshold = 400; // preset for distance
int distanceOffset = 30;

int failure = 0;

volatile int state = false;
volatile int statePrev = false;

VL53L0X sensor;
TFT_eSprite img = TFT_eSprite(&M5.Lcd);

void setup() {

  delay(2000); // wait for sensor to activate

  // M5StickC Origianl or Plus hangs if the following line is enabled.
  // Moved to link 68
  
 //   Wire.begin(32, 33, 100000UL); // (32, 33, 100000UL) for Grove socket - (0, 26, 100000UL) for HAT I2C // moved further down

 
          
    M5.begin();

  M5.Lcd.setRotation(1); // Set this to rotate the screen layout - 2 = M5 button to top. 1 = M5 button to right // added by Stuart
  M5.Lcd.setTextWrap(false);
  
    img.createSprite(160, 80);
    img.fillSprite(BLACK);
    img.setTextColor(WHITE);
    img.setTextSize(3);

    sensor.setTimeout(500);

    img.fillSprite(BLACK);
    img.setCursor(10, 10);
    img.print("Hello");
    img.pushSprite(0, 0);

    
    Serial.print(" Waiting");
    Serial.println();
    delay(5000);
    Serial.print(" And what now?");
    Serial.println();

        img.fillSprite(BLACK);
    img.setCursor(10, 10);
    img.print("Alive");
    img.pushSprite(0, 0);

    Wire.begin(32, 33, 100000UL); // (32, 33, 100000UL) for Grove socket - (0, 26, 100000UL) for HAT I2C 

    

    // Re-enable this to test the sensor connection, I think this is what is hanging the boot
//   if (!sensor.init()) {
 //     img.setCursor(10, 30);
//      img.print("Failed");
//      img.pushSprite(0, 0);
//      Serial.println("Failed to detect and initialize sensor!");
 //     while (1) {
//       }
 //   }

   while (!sensor.init()) {
     img.setCursor(10, 33);
      img.print("Fail");
           img.setCursor(20, 33);
      img.print(failure);
      failure = failure +1;
      img.pushSprite(0, 0);
      Serial.println("Failed to detect and initialize sensor!");

   }

  
    // Start continuous back-to-back mode (take readings as
    // fast as possible).  To use continuous timed mode
    // instead, provide a desired inter-measurement period in
    // ms (e.g. sensor.startContinuous(100)).
    sensor.startContinuous(300);
}

void loop() {
   uint16_t distance = sensor.readRangeContinuousMillimeters();

    Message = "Dst = " + String(distance)+"mm"; // String(distance) will convert the Int to a String for displace
//    Serial.print(Message);
    if (sensor.timeoutOccurred()) {
        Serial.print(" TIMEOUT");
    }

        if ( distance <= distanceTreshold ) {

          state = true;
           Message = ":-) " + Message;
        }
        else {
          state = false;
          Message = ":-( " + Message;
        }

        if ( state != statePrev) {
          if (state) {
          Serial.print("Detected");
            Serial.println();
            statePrev = state; 
            
        }
        
        
         if (!state) {
          Serial.print("No object Detected");
            Serial.println();
            statePrev = state;  
        }}
        

        
    img.fillSprite(BLACK);
    img.setCursor(10, 10);
    img.print(Message);
    img.pushSprite(0, 0);
}

```

_________________

Just out of curiosity, is it possible to have both the Hat and Grove ToF sensors operational?