m5stack/M5StickC #170: FactoryTest won't compile

Witchdoc59, 2023-11-10:
I have an M5Stick-C and when I try to compile FactoryTest I get this error message. 

Sketch uses 1494653 bytes (114%) of program storage space. Maximum is 1310720 bytes.
Global variables use 54248 bytes (16%) of dynamic memory, leaving 273432 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

I'm thinking that the FactoryTest is the software that came on the device when I first got it.  So it should compile and upload.  Using Arduino IDE 2.2.1 with the ESP-32 Boards 2.0.14

felmue, 2023-11-12:
Hello @Witchdoc59 

maybe this post [here](https://community.m5stack.com/topic/1797/m5stickc-cannot-upload-factorytest-size-too-large) can help?

Thanks
Felix

Cyberspy44, 2023-11-19:
Make a double check on the port and on the upload speed.